bdring/Polar-Coaster #1: Alternative Design Ideas

bdring, 2019-05-22:
It has been noted that other countries use different coaster sizes. There is a not a lot of flexibility in the current design, especially going to a larger diameter coaster.

The firmware can handle any size and probably accommodate other changes. If you are looking at a redesign, here are some thoughts.

1. The board does not need to be part of the design. A simpler existing board could be used remotely or packaged within the base. I sell some on [Tindie](https://www.tindie.com/products/33366583/pen-plotter-controller-kit-for-esp32-tmc2130/).

2. The geared rotational bed is complex to model. Someone suggested a direct drive. That is probably doable without affecting quality. I would probably suggest a [metal hub](https://www.servocity.com/770-set-screw-hubs), rather that direct mounting of the 3D printed part. It is more likely to be concentric and could buffer the heat from the motor.

fastriker, 2019-05-22:
I actually thought about a parameterized OpenSCAD model. Maybe also the gears can keep their size and an extra plate on top for the coaster lead to a constant gear transmission ratio. Then the board can also work for machines with different diameter.

bdring, 2019-05-22:
@fastriker You would need to increase the travel of the linear axis too. Maybe just move it out and increase the rod length.

dereulenspiegel, 2019-05-27:
I am actually currently trying to solve this. I increased the height of the "big gear" and added inserts for magnets. Further I modeled a platform which can hold 107mm coasters (pretty standard in Germany) which also has inserts for 3 magnets. This would allow for swappable platforms. The height of the "arm tower" is also increased by 10mm, but I probably need to increase it more.
I am currently waiting for the PCBs and other parts to try a real world assembly. If this works as expected I will publish the changes.

fastriker, 2019-05-29:
> @fastriker You would need to increase the travel of the linear axis too. Maybe just move it out and increase the rod length.

You are right - I didn't thought about that. But maybe it is also not always necessary to use the full space of the coaster.

> I am actually currently trying to solve this. I increased the height of the "big gear" and added inserts for magnets. Further I modeled a platform which can hold 107mm coasters (pretty standard in [...]

That is also a nice quick solution - and also interesting for me.