bdring/Polar-Coaster #4: BOM link for motor isn't working.

osbock, 2019-12-17:
I found a couple other Nema14x20mm steppers on that site, but don't know which one.

- 1.8 degree: https://www.omc-stepperonline.com/nema-14-stepper-motor/nema-14-bipolar-1-8deg-5ncm-7-08oz-in-0-4a-10v-35x35x20mm-4-wires.html

- 0.9 degree: https://www.omc-stepperonline.com/nema-14-stepper-motor/nema-14-bipolar-0-9deg-5ncm-7-08oz-in-0-5a-5v-35x35x20mm-4-wires.html

osbock, 2019-12-17:
Actually I looked at the original URL and it's probably this one:
https://www.omc-stepperonline.com/nema-14-stepper-motor/nema-14-bipolar-1-8deg-5ncm-7-08oz-in-0-4a-10v-35x35x20mm-4-wires.html

bdring, 2019-12-17:
That one should work. 0.9deg would work too if you double the steps/mm settings

osbock, 2019-12-18:
Two more BOM questions.
1. What is the 3mm shaft for? I can't find precut 65mm shafts, so wondering how critical end finish is.
2. I assume the belt is cut 6mm gt2 like the first version? it's not listed in the BOM

bdring, 2019-12-18:
3mm shafts are for the Pen up/down. They should be high polish finish.

Yes, cut GT2 belt